ikalogic/ScanaStudio-scripts-v3 #5: Multiple 1-wire decoder issues

EmergReanimator, 2019-11-20:
Hi fellows,

Thanks for great products so far!

Recently I upgraded ScanStudio from 3.1.0 to 4.0.7 and found out that 1-wire support is basically broken.
1. Displayed bit order is wrong (**0xf** instead of **0xf0**).
2. ROM commands are not recognized anymore (0xF0 stands for SEARCH_ROM command).
3. Decoder detects false master reset pulses (see red markers in attached screenshot.).

[image]

I  am really eager to get previous decoder back. It had only a few minor drawbacks. But still very useful. So I tried to port the decoder to new API.
Here how it looks like:
[image]

I would like to merge my changes to mainstream.

EmergReanimator, 2019-11-20:
Two more screenshots: version 0.8 vs ported one

[image]

[image]

ikalogic, 2019-12-01:
Hi @EmergReanimator 

Thank you very much for your contrinution.

You can either do a pull request or send us the ported version. In either way, we will keep full credit do you and your work!

Thank you very much again.

EmergReanimator, 2019-12-02:
Hi,

Thanks for reply!

I would like to push new branch embracing the change I made. But the I don't have permission to do that.
If you don't mind, please add me to project team. Then I could push my changes.
The state of the decoder is not yet ripe enough to be merged to master.
- Packet view is incomplete.
- Sometimes decoder misinterpret the bit stream.

So it takes a while until the changes will be available for other users.

ikalogic, 2019-12-02:
Hi, 

We have just made a test: you can:
* Fork the repo
* Apply your modification
* Create a pull request from the fork

(And no need for any permissions)

I do understand we'll need to make changes. We will review your code, and do what need to be done to make the best out of it!

Thanks!

EmergReanimator, 2019-12-08:
@ikalogic : For some reasons I observe that _on_decode_signals_ (actaully _on_decode_signals_init_) is called twice by ScanaStudio. Do you have an idea why is that?

``` javascript
function on_decode_signals(resume)
{
    if (!resume) //If resume == false, it's the first call to this function.
    {
    	on_decode_signals_init();
    }

    while ( (ScanaStudio.abort_is_requested() == false) && (ScanaStudio.trs_is_not_last(g_ch) == true) )
    {
    	on_decode_signals_decode_bit_stream();

        var tmp_trs_sample_index;
        tmp_trs_sample_index = trs.sample_index;
        while( (tmp_trs_sample_index == trs.sample_index) && (ScanaStudio.trs_is_not_last(g_ch) == true) )
        {
            TRANS = ScanaStudio.trs_get_next(g_ch);
        }
    }//end while

	while (g_owObjects.length > 0)
	{
		on_decode_signals_decode_sequence();
		// g_owObjects.pop();
	}

    return;
}
```

ikalogic, 2019-12-08:
Hi,

I have just made a quick test:

```javascript
function on_decode_signals(resume)
{
    if (!resume) //If resume == false, it's the first call to this function.
    {
        ScanaStudio.console_info_msg("Init decoder");
      
    }
  //.....
```

And the message Init decoder only appears once in my case.

What makes you think it's called twice? 

Can you share an example use case to help me reproduce this?

Thanks,

EmergReanimator, 2019-12-09:
Thanks for prompt reply!

> What makes you think it's called twice?

- In raw view I see that decoder events (e.g. MASTER RESET PULSE) are state multiple times. See attached screeshot for more details.

- Console message printed from _on_decode_signals_init()_ shows multiple times. See attached log for more details.

It might be a mistake in the 1-wire decoder script as well. But I cannot spot it. You may use attached workspace together with 1-wire decoder from EmergReanimator fork.

[image]

[1wire-searchrom.log](https://github.com/ikalogic/ScanaStudio-scripts-v3/files/3938664/1wire-searchrom.log)

[1wire-searchrom.scana.gz](https://github.com/ikalogic/ScanaStudio-scripts-v3/files/3938670/1wire-searchrom.scana.gz)

EmergReanimator, 2019-12-10:
I altered the code in such way that _on_decode_signals_ returns immediately:

```javascript
function on_decode_signals(resume)
{
	if (!resume) //If resume == false, it's the first call to this function.
    {
		ScanaStudio.console_info_msg("on_decode_signals(): !resume");
    	// on_decode_signals_init();
    }
	else
	{
		ScanaStudio.console_info_msg("on_decode_signals(): resume");
	}

	return;
}
```

And I clearly see that decoder is called multiple times depending on workspace I use 

> [Protocol decoder] on_decode_signals(): !resume
> [Protocol decoder] on_decode_signals(): !resume

ikalogic, 2019-12-10:
Okay. I was actually able to reproduce that.

This is (almost certainly) caused by the order at which elements are loaded when openning a workspace:
* The decoded is loaded first (and no signals are loaded yet). So decoding will end as soon as it starts since there are no signals to decode.
* Signal are loaded and decoder is launched again, and this time with some signals.

This should not cause any problem, and can be safely ignored. 

The order at which items are loaded by ScanaStudio, and the way it's handled, may change in the future but again, it shouldn't be causing any problem.

ikalogic, 2019-12-10:
By the way, if you refresh the decoder:
[image]
you will see the initialization is only executed once

ikalogic, 2019-12-10:
Oh, and one last thing, looking at your screenshot, i see there are 2 1-wire scripts running, hence the double items in the raw view :)

[image]

EmergReanimator, 2019-12-11:
> By the way, if you refresh the decoder:
> [image]
> you will see the initialization is only executed once

Yes, it works the same for me. Single message appears as soon as I press refresh button.

EmergReanimator, 2019-12-11:
> Oh, and one last thing, looking at your screenshot, i see there are 2 1-wire scripts running, hence the double items in the raw view :)
> 
> [image]

I tried with one decoder only. But it does really matter. Two messages still show up. The odd thing about these messages, that sometimes I got a single message and another time there are 2 or more of the them. Seems to be a timing issue to me.

[image]

ikalogic, 2019-12-20:
Hi EmergReanimator

I'm currently working on fixing the issues with the 1-Wire, at least the most critical parts.

I would love to merge your changes to main stream, when do you think we could do that?

Thanks a lot!

EmergReanimator, 2019-12-21:
Hi,

In fact, the decoder itself works quite fine. Builder is still not ready.
But major issue now is decoding on mixed standard and overdrive speeds. This will requires major changes: bit speed depends on decoding.
I need some time to clea-up the code. So I believe in couple of days is it will be ready to be merged.

ikalogic, 2019-12-21:
That's impressive :) 👍 

Thank you very much for your contribution

ikalogic, 2019-12-23:
@EmergReanimator We (Ikalogic) would like to offer you an SP209 device, could you reach to us by mail at contact@ikalogic.com ?

commit 302a5e6c (2020-01-08): [OE] Corrected file header. Closes #5
closed 2020-01-08 by ikalogic