ikalogic/ScanaStudio-scripts-v3 #14: Scanastudio script manager duplicates some protocols in  4.0.15

YoannD42, 2020-07-17:
In its last version, 4.0.15, running on a Windows 10 (1909 Enterprise) 64bits machine, scanastudio script manager will duplicate some protocols : 

- 1-Wire is displayed 3 times. 
- SPI Flash memory is displayed 2 times. 

Issue persists after clicking refresh, or a ScanaStudio restart. I can't reproduce the issue under linux (debian 64 bits).

[image]
[image]

YoannD42, 2020-07-17:
In addition, I just sent the application and session logs, with a link to this Issue, if it can help to reproduce. 

Thank you !

BastitNi, 2020-07-17:
Hi @YoannD42,
With the log file you just sent, it appears that the duplicated script were made "manually". Their names are *protocolname_copy0.js*

Thus, you should remove them manually, because ScanaStudio must not delete a script written by a user.

YoannD42, 2020-07-17:
Hi, thank you for the fast answer, 

For the _copy0 I agree, that's my copies, that's the 2 scripts you see in the screenshots, named "SPI GTX DAC" (this one is a copy from spi) and "STIM DECODER" (this one is copied from pwm). 

For the rest, 1-Wire and Spi Flash protocols, it looks like... different ? The 2 copies were made at the same timedate, and not names _copyx,  could it be after an update ? 

[image]

BastitNi, 2020-07-17:
Thank you for your feedback. I'll investigate around that.
Unfortunately, ScanaStudio won't be able to remove these script as I said before, but it should prevent their creation.
It will probably be fixed in the next release.

YoannD42, 2020-07-17:
Perfect, Thanks for taking feedback.

closed 2020-07-17 by YoannD42