ikalogic/ScanaStudio-scripts-v3 #18: Support for PDM data

jonnor, 2020-11-20:
[Pulse Density Modulation PDM](https://en.wikipedia.org/wiki/Pulse-density_modulation) is common on digital microphones. Would be a good addition to compliment the I2S support that is already available.

ikalogic, 2023-01-20:
Hi,

I know it have been two year, but better late than never.

Are you able, by any chance, to capture some sample PDM signals?

I'm not familiar with the PDM protocol, but can gladly try and implement a decoder/generator for that.

jonnor, 2023-01-20:
Hi and thank you for the response! I am not working so actively with PDM anymore, so unfortunately I am not able to provide any samples at this time. At the time of filing the issue we were debugging problems with PDM microphone, but we got that sorted out eventually[1]. So unless anyone else really wants/needs this, maybe the issue can be closed.

1. Turned out the clock was an unsupported configuration for the particular mic, right between low-power and performance modes, so tiny changes in environment / between units it would make it flip between the modes and have odd dropouts in the transitions.

ikalogic, 2023-01-20:
Glad you got your project working.

I've been doing some reading (PDM is a way more sophisticated subject that i thought!), and i think it would be nice to have it in ScanaStudio library.

I have a question though: is PCM considered synchronous or asynchronous? Obviously there is a clock in the system, but can we assume it's always available and part of the signals being captured? Or PDM can exist as a single wire, without any clock signal?

Depending on your answer, I'll have to think about how far decoding goes (how many filters are applied). 

My idea, as a user of the logic analyzer, is that the more sophisticated is the applied filtering, the further it will be from captured data, and the harder it will be to debug anything. In other words, i think applying a moving average is enough as a post processing. What do you think? You opinion is very precious as you had to debug such a system!

Thank you very much,

jonnor, 2023-01-29:
Physically I think PDM always has to have the clock. But in practice, if one would observe only the data signal *and* also know which clock (in kHz/Mhz) and oversampling rate (typically 64x), then I think that is all that is needed in order to convert to PCM (audio samples, with audio sample rate typ in 8-96khz range).

I think it would be fine to have to specify the clock and oversampling rate, if that simplifies anything.

I think a very basic PDM to PCM conversion would be excellent (moving average etc). Certainly enough to debug the kind of issues we had!