ikalogic/ScanaStudio-scripts-v3 #22: The csv import script does not import data into the generator.

rigelinorion, 2023-03-21:
Hello. I cannot import data from a file generated from a waveform record. First line skipped, column break tag is ";", column0 as time, column1 imported into channel1. I have no idea anymore. I am attaching a snippet of the file that I am trying to import data.
My device is a ScanaQuad SQ200. The software is ScanaQuad 5.1.2.

Time;CH 1 1-Wire
0,0050707160;1
0,0056057160;0
0,0056081400;1
0,0056090160;0
0,0056114440;1
0,0056123040;0
0,0056147320;1
0,0056155920;0
0,0056180200;1
0,0056188800;0
0,0056213080;1
0,0056221680;0
0,0056245960;1
0,0056254560;0
0,0056278840;1
0,0056287440;0
0,0056311720;1
0,0056320320;0
0,0056344600;1
0,0056353400;0
0,0056377680;1
0,0056386240;0
0,0056410560;1
0,0056419120;0

rigelinorion, 2023-03-21:
I found a bug. The csv_structure variable, when the Select CSV structure option is set to "1 or more transitions per line", always holds the value 0, i.e. "1 sample per line". After manually setting it to 1 in the code, everything works, signal builder loads the waveform without a problem. It seems to me that the ScanaLogic program has some problem with the correct interpretation of the script. In addition, after running the script, its name does not appear in the Signal Builder tab. It appears only after opening the settings window and clicking apply. Weird.

ikalogic, 2023-03-21:
Dear @rigelinorion ,

Thanks for the feedback, we have openned an internal ticket and we'll be fixing this ASAP.

Thank you very much.

CC @BastitNi

rigelinorion, 2023-03-28:
Thanks for the update(5.1.3). Now it works fine.

closed 2023-03-28 by rigelinorion